NabuCasa/silabs-firmware-builder #1: [REQUEST] Add device branding using userdata token with manufacturer and board name via a tokenfile?

Hedda, 2023-02-10:
@agners  Maybe consider adding unique device branding using userdata token with manufacturer and board name via a tokenfile?

@MattWestb has posted instructions here -> https://github.com/MattWestb/EFR32-FW/blob/main/Branding_EFR32/README.md

Copy of his readme below:

All EFR32 chips have one dictated aria in the flash for user data named `userdata`.  
Its stors some mandufactur tokens and also cunstom tokens.  
Its not being erased of debug unlock / flash erase andd firmware flashing. Chip ease and writing "erase" as token data is deleting it or writing it new updated token.   
IKEA is using it to storing cuntom token with overriding the mdele ID and other parameters that is in the firmware for using the same firmware for mor products like the 5 buttom remotes have the same firmware but diferent names.  

For NCP firmware you can storing some of the tokens and bellows / ZHA is readding them and you iss getting in the log.  

You can writing token one by one or more from one file.  
You can writing them to chip (flashing the chip with new tokens) or appanding them to one firmware that is flasing the firmware and the new tokens.  

Example of tocken file:
```
MFG_STRING                    : "IKEA of Sweden"
MFG_BOARD_NAME                : "Billy EZSP by MW"
MFG_MANUF_ID                  : 0x110C
```
And flashing the token file with simplicity commander:
```
.\commander.exe flash --tokengroup znet --tokenfile .\tokens-MW.txt -d EFR32MG1P132F256IM32
WARNING: The string for token MFG_BOARD_NAME (Billy EZSP by MW) fills all 16 bytes of the token space. The string will not be zero terminated.
Writing 2048 bytes starting at address 0x0fe00000
Comparing range 0x0FE00000 - 0x0FE007FF (2 KB)
Erasing range 0x0FE00000 - 0x0FE007FF (1 sector, 2 KB)
Programming range 0x0FE00000 - 0x0FE001FF (512 Bytes)
Programming range 0x0FE00200 - 0x0FE003FF (512 Bytes)
DONE
```
You is getting this from bellows info:
```
Manufacturer: IKEA of Sweden
Board name: Billy EZSP by MW
EmberZNet version: 6.9.0.0 build 178

```
The same shall working with this commands for appending the token to one firmware file:
>$ commander convert blink.s37 --tokengroup znet --tokenfile tokens.txt --device EFR32MG1P --outfile blink.hex Converts blink.s37 to hex format, while simultaneously defining the tokens defined in tokens.txt and on the command line.

Only putting your parameters for your chip and the in and output file.  

Editing: Its looks its only possible using the "flash" or "convert" s37 to s37 fils and adding the tokens but to GBL is not working and if having one s37 file with tokens and converting / craating one GBL its only taking the app and not the extra data for the tokens.
I think that is made for not being possible "streeling" devices without having hw access to them (= SWD flashing).

Alos if you is having one IKEA device hockedup you shall dumping the user airia that can being used for "transforming" one device if you like.  
The main firmware can beeing restored from one OTA filee but not the user data so all dumps of its is wery wellcome in the comunity for firmware transforming.

MattWestb, 2023-02-10:
SC that is shipped is already branded with manufacture token in user data.
```
Manufacturer: Nabu Casa
Board name: SkyConnect v1.0
EmberZNet version: 7.0.2.0 build 406
```

Hedda, 2023-02-10:
> SC that is shipped is already branded with manufacture token in user data.

So should writing of that userdata token be separate from this https://github.com/NabuCasa/silabs-firmware-builder project or?

@puddly If so would this request be better suited for the https://github.com/NabuCasa/universal-silabs-flasher project instead?

MattWestb, 2023-02-10:
The token **cant** being flashed with GBL file (at least documented by SL) its flashed in the manufacturing processes and cant being updated after it left the factory so its out of topic.

MattWestb, 2023-02-10:
@Hedda Pleas read https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf !!

MattWestb, 2023-02-10:
And https://www.silabs.com/documents/public/application-notes/an961-custom-nodes-efr32.pdf.

agners, 2023-03-14:
> The token **cant** being flashed with GBL file (at least documented by SL)

It can be flashed through the SE manager API, and there is an example application in the Gecko SDK, see [se_manager_user_data](https://github.com/SiliconLabs/gecko_sdk/tree/gsdk_4.2/app/common/example/se_manager_user_data) (specifically `app_process.c`).

> its flashed in the manufacturing processes and cant being updated after it left the factory so its out of topic.
> 
> = have nothing to do with the firmware building and flashing GBL files from the bootloader.

Yes agreed, token flashing is out-of-topic for this repository. There are more Manufacturing issues than token flashing, and our goal is not to create a set of tools to ease manufacturing. SiLab already provides much of that.

closed 2023-03-14 by agners

MattWestb, 2023-03-14:
I think one good download for getting most interesting docks for Silabs Zigbee and thread products is: https://github.com/SiliconLabs/gecko_sdk/releases/download/v4.2.2/documentation.zip and its take little time reading and little longer for understanding (if can do it).
So pleas @Hedda go and getting ti !!