NabuCasa/silabs-firmware-builder #15: Board Activity LED

darkxst, 2023-04-19:
How does the activity LED work on the Yellow? Is there some code running on the Yellow that drives this LED? or is it built into firmware to display activity?

agners, 2023-05-04:
It is built into the firmware. The Zigbee firmware has support for it, the Multi-PAN as well as the OpenThread RCP firmware do not have support at this point (or rather, I did not find something built-in, probably needs custom code).

MattWestb, 2023-06-10:
GIPO control is implanted in GSDK 4.3.0.0 but is not in stable / production status and need being implanted and tested before if someone is interested doing the code work.

darkxst, 2023-06-10:
Yes noticed that in the release notes

> Added the CPC GPIO Expander component, a valuable addition that empowers users to discover and harness the GPIO pins available on the device. With this feature, the Host system gains effortless access to the GPIO capabilities provided by the CPC device. The CPC GPIO Expander component streamlines the process of exposing and utilizing GPIO pins. On a Linux Host, a dedicated kernel driver can be installed, enabling seamless access to the GPIOs present on the device, treating them as standard GPIOs. This integration allows users to interact with the CPC GPIOs using familiar Linux GPIO APIs, simplifying the development and integration of GPIO-based functionalities.

agners, 2023-07-10:
> On a Linux Host, a dedicated kernel driver can be installed, enabling seamless access to the GPIOs present on the device, treating them as standard GPIOs.

So this is meant for a host component to control a LED.

Not sure if `zigbeed` or their OpenThread fork can then use that GPIO.

I wonder why can't the Multi-PAN RCP firmware not control the LED directly, on RF activity? IMHO, since we only have a single LED, this would be the best and easiest way to control the LED. Otherwise OpenThread/Zigbeed would need to share a single LED which might be problematic even.

darkxst, 2023-08-10:
not stale

darkxst, 2023-08-12:
The activity LED is stubbed out in `zigbeed` build. referenced code is in a binary blob I guess, but clearly its not triggering on RF activity. perhaps could hook into cpc to achieve the same.
```
// Referenced from sli_zigbee_network_check_incoming_queue in network.c
// Stub in platform/base/hal/micro/generic/led-stub.c
void halStackIndicateActivity(bool turnOn)
{
}

MattWestb, 2023-09-22:
One interesting function but its not what you is looking for:
```
CPC GPIO Expander
Description
The Co-Processor Communication (CPC) General Purpose Input/Output (GPIO) Expander 
is a software component designed to enable a CPC Primary/Host device to utilize a 
Secondary  device's GPIOs as if they were its own. With the CPC GPIO Expander, the 
Primary/Host  device can seamlessly integrate with the Secondary device and make use 
of its GPIO  capabilities. This component is dependent on specific pieces of software on 
the  Primary/Host device to facilitate this communication. By enabling this communication,  
the CPC GPIO Expander expands the capabilities of the Primary/Host device, allowing  
it to interface with a wider range of devices and peripherals.

Quality PRODUCTION
```

Hedda, 2024-06-20:
FYI, Zigbee EmberZNet firmware shipping with ITead's Sonoff ZBDongle-E expose control for brightness of the indicator light LED:

https://www.zigbee2mqtt.io/devices/ZBDongle-E.html#light-indicator-level-numeric

### Light indicator level (numeric)
Brightness of the indicator light.
Value can be found in the published state on the `light_indicator_level` property.
It's not possible to read (`/get`) or write (`/set`) this value.

darkxst, 2024-06-20:
> firmware shipping with ITead's Sonoff ZBDongle-E expose control for brightness of the indicator light LED:

Only their router firmware exposes the led. That is somewhat different as its just a standard Zigbee cluster.

puddly, 2025-12-09:
https://github.com/NabuCasa/silabs-firmware-builder/pull/158 implements this for newer SDKs.