NabuCasa/silabs-firmware-builder #25: Read permissions broken on Docker image

darkxst, 2023-06-21:
This has been broken for sometime, can we get the read permissions updated on the this? Currently its not possible for the end-user to pull these images (as per instructions in README.md)
```
docker run --rm -it \
  --user builder \
  -v $(pwd):/build -v ~/.gitconfig:/home/builder/.gitconfig \
  ghcr.io/nabucasa/silabs-firmware-builder:4.3.0
Unable to find image 'ghcr.io/nabucasa/silabs-firmware-builder:4.3.0' locally
docker: Error response from daemon: Head "https://ghcr.io/v2/nabucasa/silabs-firmware-builder/manifests/4.3.0": unauthorized.
```

caipiblack, 2023-07-08:
Same problem

agners, 2023-07-10:
The image is public now. Thanks for bringing this up!

closed 2023-07-10 by agners