NabuCasa/silabs-firmware-builder #40: Missing paramerization in builder

wolf99, 2023-12-27:
https://github.com/NabuCasa/silabs-firmware-builder/blame/3d23e240b62137304397841d64ceecf6baddd9e7/Dockerfile#L44

Just taking a quick glance through the files here ot of curiosity so I could well be wrong, but it seems like this line should, like the others, refernce the version variable rather than hard code it?

Just a product care nit for future proofing.

Haven't submitted a pull request as I have no good setup to test changes.

darkxst, 2023-12-27:
These links will need to be changed for Gecko 4.4.0 anyway as that now needs gcc 12.2

darkxst, 2023-12-28:
Just for reference
https://github.com/darkxst/silabs-firmware-builder/commit/058088436df3eb58206ac4b83fbe86331616f503

wolf99, 2023-12-28:
My initial curiosity was about the tool chain used and I noticed keil mentioned. Is that something that is specific to the silabs chip or perhaps just what the maintainers happen to have available?

If not specific, then is it technically feasible to change to some FOSS tool(s)?

puddly, 2023-12-28:
> If not specific, then is it technically feasible to change to some FOSS tool(s)?

I think the `armkeli` reference in the URL is just the account name for Azure Blob Storage. We're replicating the toolchain and build process of the IDE, which uses GCC + binutils, etc.

> [Arm GNU Toolchain](https://developer.arm.com/Tools%20and%20Software/GNU%20Toolchain) is a community supported pre-built GNU compiler toolchain for Arm based CPUs.

darkxst, 2023-12-29:
Silicon Labs support two different toolchains the open source GNU toolchain (that we are using) or the propriety IAR toolchain. Both toolchains are generic to ARM and not specific to Silabs chips.

agners, 2024-01-03:
This is addressed with #42.

closed 2024-01-03 by agners