NabuCasa/silabs-firmware-builder #47: Flashing my SiliconLabs BV 2010/10 with the firmwares from here?

IngwiePhoenix, 2024-02-11:
Hello there!

While trying to configure my stick with Home Assistant, I ended up looking around for possible reasons why it did not work. Turns out that I have to flash a new firmware to it.

`lsusb` reports it as such:
```
Bus 001 Device 003: ID 10c4:8b34 Silicon Labs BV 2010/10
```

Since I run Home Assistant Core, I have to do these things manually and can not rely on an automation add-on to do it for me, unfortunately.

Which build do I have to apply to the listed steps in the README to generate and subsequently flash the proper firmware to my device?

Thanks and kind regards,
Ingwie

puddly, 2024-02-11:
Your stick is based on the EM3587 chip, which isn't supported by newer SDKs. This repository only contains firmware compatible with newer chips like the EFR32.

Your specific stick seems to do something strange with serial ports and already has (old) firmware installed.

closed 2024-02-11 by puddly

darkxst, 2024-02-11:
> compatible with newer chips like the EFR32

Really requires EFR32 Series 2 chips