NabuCasa/silabs-firmware-builder #48: Can this work on other SiLabs devices?

IIAIronWolf, 2024-02-12:
I have the NRF52840. It was hard enough getting it flashed to run OpenThread. It would be awesome if this could be used to program other SiLabs devices, specifically the NRF52840, as it is a suggested board in the OpenThread Border Router documentation in the HA add-ons.

darkxst, 2024-02-12:
> specifically the NRF52840

That would be Nordic, not Silabs. 

This builder is very tightly coupled to the Silabs Gecko SDK, and cant be used to build other manufacturers SDK's.

closed 2024-03-16 by puddly

IIAIronWolf, 2024-04-28:
> > specifically the NRF52840
> 
> 
> 
> That would be Nordic, not Silabs. 
> 
> 
> 
> This builder is very tightly coupled to the Silabs Gecko SDK, and cant be used to build other manufacturers SDK's. 

Regardless, why would HA put it in their documentation as an option (NRF52840). I literally only bought it because it was suggested.

puddly, 2024-04-28:
What specific documentation are you referring to?

IIAIronWolf, 2024-04-30:
> What specific documentation are you referring to?

The OTBR documentation. [image]

agners, 2024-04-30:
> Regardless, why would HA put it in their documentation as an option (NRF52840). I literally only bought it because it was suggested.

The OpenThread Border Router add-on can talk to any OpenThread RCP radio talking the Spinel protocol. It should work with the nRF52840 radios as well, I've tested that radio in the past already.

However, this is all unrelated to this particular GitHub repository: This is about firmware builds for SiLabs radio.

As the documentation says, there is a easy route with SkyConnect/ZBT-1 and Yellow since the add-on pre-installs it. For alternative radio you are on your own. But there is article linked.

In any case, this is unrelated to this repository. If you belive that you flashed a correct firmware, and still have issues running the OpenThread Border Router add-on, please open an issue at https://github.com/home-assistant/addons along with the add-on logs when using your radio and information what firmware and the process you used to flash the firmware for it.

IIAIronWolf, 2024-05-17:
> > Regardless, why would HA put it in their documentation as an option (NRF52840). I literally only bought it because it was suggested.
> 
> 
> 
> The OpenThread Border Router add-on can talk to any OpenThread RCP radio talking the Spinel protocol. It should work with the nRF52840 radios as well, in fact, I've tested that radio in the past already.
> 
> 
> 
> However, this is all unrelated to this particular GitHub repository: This is about firmware builds for SiLabs radio.
> 
> 
> 
> As the documentation says, there is a easy route with SkyConnect/ZBT-1 and Yellow since the add-on pre-installs it. For alternative radio you are on your own. But there is article linked.
> 
> 
> 
> In any case, this is unrelated to this repository. If you belive that you flashed a correct firmware, and still have issues running the OpenThread Border Router add-on, please open an issue at https://github.com/home-assistant/addons along with the add-on logs when using your radio and information what firmware and the process you used to flash the firmware for it.
> 
> 
> 
> 
> 
> 

I went ahead and bought a skyconnect, but I still think the OTBR documentation should be changed. It says to 'use this article' for setting up the nRF, but that article no longer exists.