NabuCasa/silabs-firmware-builder #51: Bricked SkyConnect

cvhoang, 2024-03-16:
Hi there,

There seems to be an issue with the build when you change the configuration of EZSP build. The resulted firmware will brick SkyConnect. For example, in the following configuration, I increase key table size to 100 and reduce router table size to 100, SkyConnect simply doesn't boot up. You can connect to the bootloader and the stick is bricked.

```
  ezsp-firmware-build:
    name: EmberZNet NCP
    needs: build-container
    strategy:
      matrix:
        include:
          - target: yellow
            device: MGM210PA32JIA
            components: simple_led:board_activity,zigbee_token_interface,zigbee_mfglib
            patchpath: "EmberZNet/Yellow"
          - target: skyconnect
            device: EFR32MG21A020F512IM32
            components: zigbee_token_interface,zigbee_mfglib
            patchpath: "EmberZNet/SkyConnect"
    uses: ./.github/workflows/silabs-firmware-build.yaml
    with:
      image_name: ${{ needs.build-container.outputs.image_name }}
      firmware_name: ncp-uart-hw-${{ matrix.target }}
      project_file: "/gecko_sdk/protocol/zigbee/app/ncp/sample-app/ncp-uart-hw/ncp-uart-hw.slcp"
      project_name: "ncp-uart-hw"
      device: ${{ matrix.device }}
      components: ${{ matrix.components }}
      configuration:  |
        SL_IOSTREAM_USART_VCOM_RX_BUFFER_SIZE:64,
        EMBER_APS_UNICAST_MESSAGE_COUNT:20,
        EMBER_NEIGHBOR_TABLE_SIZE:26,
        EMBER_SOURCE_ROUTE_TABLE_SIZE:100,
        EMBER_KEY_TABLE_SIZE:100,
      patchpath: ${{ matrix.patchpath }}
      extra_c_defs: "'-DEMBER_ADDRESS_TABLE_SIZE=16' '-DEMBER_MULTICAST_TABLE_SIZE=16'"
      sdk_version: ${{ needs.build-container.outputs.sdk_version }}
      metadata_fw_type: "ncp-uart-hw"
      baudrate: 115200
      metadata_extra: "{ \"ezsp_version\": \"7.4.0.0\" }"
```

Any idea why? I need to increase the key table size to do migration from Zigbee2MQTT. Otherwise, only 12 devices will move across. I have 96 devices to migrate. Is there away to "unbricked" SkyConnect?

puddly, 2024-03-16:
How are you performing the migration? You really don't need to tweak anything to use the SkyConnect, especially like this. People have hundreds of devices connected using the default values.

If you can connect to the bootloader, you can flash back the original firmware with https://github.com/nabucasa/universal-silabs-flasher/. If not, you have to flash it with an SWD debugger via the debug pads.

cvhoang, 2024-03-17:
I use `zigpy-cli` to back up CC2652P dongle and restore it to SkyConnect. I'm actually using Zigbee2MQTT. If I use the default `EMBER_KEY_TABLE_SIZE=12`, there are only 12 devices migrated across to SkyConnect. That's why I was trying to increase this value.

Unfortunately I can't seem to seem to connect the stick using `universal-silabs-flasher`.

```
2024-03-16 17:39:12.093 westeros asyncio DEBUG Using selector: EpollSelector
2024-03-16 17:39:12.094 westeros universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.0', ezsp_version='7.4.0.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-u
art-hw'>, baudrate=115200)
2024-03-16 17:39:12.095 westeros universal_silabs_flasher.flash DEBUG Probing app type ApplicationType.EZSP first
2024-03-16 17:39:12.095 westeros universal_silabs_flasher.flash DEBUG Probing with 115200 baudrate first
2024-03-16 17:39:12.095 westeros universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-03-16 17:39:12.095 westeros zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB1' (115200 baudrate)
2024-03-16 17:39:12.097 westeros universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x7fea23acd0a0>
, Serial<id=0x7fea23b2adc0, open=True>(port='/dev/ttyUSB1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-03-16 17:39:12.098 westeros universal_silabs_flasher.common DEBUG Sending data b'\n'
2024-03-16 17:39:12.098 westeros universal_silabs_flasher.common DEBUG Sending data b'3'
2024-03-16 17:39:14.101 westeros universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-03-16 17:39:14.102 westeros zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB1' (115200 baudrate)
2024-03-16 17:39:14.104 westeros bellows.ezsp DEBUG Resetting EZSP
2024-03-16 17:39:14.105 westeros bellows.uart DEBUG Resetting ASH
2024-03-16 17:39:14.105 westeros bellows.uart DEBUG Sending: b'1ac038bc7e'
2024-03-16 17:39:19.110 westeros universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2024-03-16 17:39:19.110 westeros zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB1' (460800 baudrate)
2024-03-16 17:39:19.113 westeros bellows.uart DEBUG Connection lost: None
2024-03-16 17:39:19.113 westeros bellows.uart DEBUG Closed serial connection
```

puddly, 2024-03-17:
> there are only 12 devices migrated across to SkyConnect. 

Where do you get this number from?

Keep in mind that 99% of the devices on your Zigbee network will not show up in either network backup.

cvhoang, 2024-03-17:
Once the backup was restored to SkyConnect, I did a backup for SkyConnect itself and inspected the generated json file. There are only the first 12 devices in the backup of the old coordinator in this file. If you do the same migration using Home Assistant method (https://skyconnect.home-assistant.io/migrate-zigbee-zha/), the result is the same. In Home Assistant log, there will be multiple warning "Can't add key due to Ember.TABLE_FULL errors" or something similar.

puddly, 2024-03-17:
They're warnings specifically because they aren't a critical problem: APS link keys aren't used by most devices. Even if the key exists, it isn't necessary to communicate with the device.

cvhoang, 2024-03-17:
Oh OK. I inspected the `devices` section of SkyConnect's backup and there were only 12 of them. That's why I thought only 12 devices were migrated. I guess this list will grow once more devices talk to the new coordinator?

puddly, 2024-03-17:
It'll only contain information about devices that have useful state. If you have a device that just sits on the network without extra APS encryption keys or other useful metadata, it won't exist in the backup because there's no new info to be written to the coordinator when restoring. The device will continue to work as before.

cvhoang, 2024-03-17:
Thank you for your help!

closed 2024-04-23 by github-actions[bot]