NabuCasa/silabs-firmware-builder #75: Intermittent build failures

darkxst, 2024-06-29:
I am testing out the reworked builder and seeing this intermittent failure, If I rebuild the failed workflow it usually works the second time.

```
INFO:__main__:Version is correct, picking /opt/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi
Slcp parse warnings for /__w/silabs-firmware-builder/silabs-firmware-builder/build/rcp-uart-802154.slcp
	Referenced SDK Extension 'nc_efr32_watchdog:1.0.0' could not be found.
	Referenced project component nc_efr32_watchdog from extension 'nc_efr32_watchdog' not in framework.
Warnings loading project: 
 - Referenced SDK Extension 'nc_efr32_watchdog:1.0.0' could not be found.
 - Referenced project component nc_efr32_watchdog from extension 'nc_efr32_watchdog' not in framework.
Found valid project type for makefile
Found valid toolchain component toolchain_gcc for toolchain_gcc
Generation failed: /__w/silabs-firmware-builder/silabs-firmware-builder/build/rcp-uart-8021[54](https://github.com/tl-sl/silabs-firmware-builder/actions/runs/9720459888/job/26831876765#step:5:55).slcp contains parse issues. As of .slcs spec version 8, projects must not contain any warnings to generate.
To override this behaviour, opt in using --force.
Traceback (most recent call last):
  File "/__w/silabs-firmware-builder/silabs-firmware-builder/tools/build_project.py", line 648, in <module>
    main()
  File "/__w/silabs-firmware-builder/silabs-firmware-builder/tools/build_project.py", line 469, in main
    subprocess.run(
  File "/usr/lib/python3.11/subprocess.py", line [57](https://github.com/tl-sl/silabs-firmware-builder/actions/runs/9720459888/job/26831876765#step:5:58)1, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['slc', '--daemon', '--daemon-timeout', '1', 'generate', '--project-file', PosixPath('/__w/silabs-firmware-builder/silabs-firmware-builder/build/rcp-uart-802154.slcp'), '--export-destination', PosixPath('/__w/silabs-firmware-builder/silabs-firmware-builder/build'), '--sdk', PosixPath('/gecko_sdk_4.4.2'), '--toolchain', 'toolchain_gcc', '--output-type', 'makefile']' returned non-zero exit status 1.
```

puddly, 2024-06-30:
I think I've run into this once but it's hard to tell why this is happening. Perhaps it's due to the daemonization of `slc`.

darkxst, 2024-10-23:
Not seen this again recently.

closed 2024-10-23 by darkxst