NabuCasa/silabs-firmware-builder #81: build_project.py broken under Windows.

rnicolas, 2024-09-25:
Just follow the steps described in the README. When trying to build the firmware, I have got this error:

```
INFO:__main__:Building in C:\Users\Dell\SimplicityStudio\v5_workspace\silabs-firmware-builder\build
INFO:__main__:SDK ..\..\SDKs\gecko_sdk_4.4.4 has version 4.4.4
INFO:__main__:Version is correct, picking ..\..\SDKs\gecko_sdk_4.4.4
INFO:__main__:Building component graph and identifying board-specific files
Traceback (most recent call last):
  File "C:\Users\Dell\SimplicityStudio\v5_workspace\silabs-firmware-builder\tools\build_project.py", line 648, in <module>
    main()
  File "C:\Users\Dell\SimplicityStudio\v5_workspace\silabs-firmware-builder\tools\build_project.py", line 416, in main
    for name in determine_chip_specific_config_filenames(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dell\SimplicityStudio\v5_workspace\silabs-firmware-builder\tools\build_project.py", line 147, in determine_chip_specific_config_filenames
    proc = subprocess.run(
           ^^^^^^^^^^^^^^^
  File "C:\Users\Dell\AppData\Local\Programs\Python\Python312\Lib\subprocess.py", line 548, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dell\AppData\Local\Programs\Python\Python312\Lib\subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Dell\AppData\Local\Programs\Python\Python312\Lib\subprocess.py", line 1538, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

The command called is:

```
python .\tools\build_project.py --sdk ..\..\SDKs\gecko_sdk_4.4.4 --toolchain C:\SiliconLabs\SimplicityStudio\v5\developer\toolchains\gnu_arm\12.2.rel1_2023.7 --manifest .\manifests\nabucasa\yellow_ncp-uart-hw.yaml --build-dir build --output-dir output --output gbl --output out --output hex
```

rnicolas, 2024-09-25:
Found what triggered that error, turns out that even if slc is installed and in the path, the script needs to specify that the command to be executed is `slc.bat` in line 38. Now it's blaming it cannot run `make`, but I cannot find any make binary in the Simplicity Studio folders. I see in the `Dockerfile` it is installed as a dependency, but that is not specified in the `README.md`

puddly, 2024-09-25:
Windows isn't really a supported platform. We rely on GNU Make to build the resulting project so if you don't have it, either install it or run the tools via WSL.

rnicolas, 2024-09-25:
I managed to get something functional, but instead of using WSL and reinstalling all the tools I had already set up, I used MSYS2 and added its binaries to the system path. This allowed me to run the build script up to the point where it calls the script to generate the GBL file, which fails due to some improperly escaped paths. However, when I manually run the GBL generation script, it successfully creates a GBL file that is recognized by the Universal Silabs Flasher and also by the Home Assistant configurator when flashed. Now, I just need to figure out what's causing the issue with the incorrect paths.

rnicolas, 2024-09-27:
I wanted to share how I managed in the end. And maybe it helps as reference for someone that has the system already setup in Windows and does not want to mix with more tools.

I installed MSYS2 by issuing `winget install MSYS2.MSYS2`, then in the MSYS2 terminal I installed `make` by executing `pacman -S make`, and then adding `C:\msys64\usr\bin` to the path.

After that, I made some adjustments in `build_project.py`, I successfully got everything working "nicely." The main issue I encountered was the incorrect recognition by `make` when trying to run Python. To resolve this, I removed the postbuild injection to the makefile and instead I called create_gbl.py right after the make command.

Here’s the revised post-build command I implemented:

```
    # Postbuild command
    command = [
        "python",
        str(args.postbuild).replace("\\", "/").lstrip("C:"),
        "postbuild",
        str((args.build_dir / base_project_name).with_suffix(".slpb").resolve()).replace("\\", "/").lstrip("C:"),
        "--parameter", "build_dir:" + str(output_artifact.parent.resolve()).replace("\\", "/").lstrip("C:"),
        "--parameter", "sdk_dir:" + str(sdk).replace("\\", "/").lstrip("C:")
    ]

    # Run the command using subprocess
    result = subprocess.run(command, check=True, text=True)
```

The command I am using to run it is:

`python "tools/build_project.py" --sdk "C:/Users/Dell/SimplicityStudio/SDKs/gecko_sdk_4.4.4" --toolchain "C:/SiliconLabs/SimplicityStudio/v5/developer/toolchains/gnu_arm/12.2.rel1_2023.7" --manifest "C:/Users/Dell/SimplicityStudio/v5_workspace/silabs-firmware-builder/manifests/nabucasa/yellow_ncp-uart-hw.yaml" --postbuild "C:/Users/Dell/SimplicityStudio/v5_workspace/silabs-firmware-builder/tools/create_gbl.py" --build-dir build --output-dir output --output gbl --output out --output hex`

And that works smooth. Once that said, this is more like a workaround that a proper solution to make it running in MS Windows. So whoever is in charge of issues, do with this information whatever you want.