NabuCasa/silabs-firmware-builder #96: Help needed - MG21A010HI part of ISP router - Arcadyan

kar200, 2024-12-16:
Hi,

First of all I apologize for hijacking here if this is not relevant. I am just after some guidance on where to get help please.

I have an ISP router provided by the ISP. It is based on Broadcom bcmbca (BCM68360_B1).

The firmware doesn't do much as expected and does not contain any zigbee related config. I noticed that it has a zigbee chip (MG21A010HI). The only device I found online with a similar chip is the Homey pro with no access to app files.

The firmware does contain tools to flash GBL files directly. It also contains 3 GBL files for DMP, Zigbee and BLE. Flashing them is straight forward.

I can run any linux distro based on aarch64 in chroot. everything works fine and I was trying to check if I can run zigbee2mqtt.
After failing I noticed that any tool I try and run with any of the 3 firmware does not allow any communication. The silabs firmware flasher just says that there is no communication or is in bootloader mode. 

I tried a few dozen GBL files and none of them work. I don't get anything on the CLI (not sure if I am supposed to) and it does not work with any of the silabs python projects I tried to communicate with the chip (at least probe).

The serial port is /dev/ttyH0. 

I am attaching a picture of the chip (excuse the skill it's the best I could do). I can also attach the GBL files or converted to S37 format if that can help with anything.

and by the way the included gbl files work well in cli mode. I have not tested much.

Below is a sample
````
~ # dmp
Zigbee Network Status:

Short ID: 0x738B, EUI64: (>)9035EAFFFEA1EE64, Pan ID: 0x3EBE
Z3LightSoc_PA_010>info
MFG String: 
AppBuilder MFG Code: 0x1002
node [(>)9035EAFFFEA1EE64] chan [11] pwr [3]
panID [0x3EBE] nodeID [0x738B] xpan [0x(>)F1CC66331C3F9A7C]
parentID [0xFFFF] parentRssi [0]
stack ver. [6.9.0 GA build 178]
nodeType [0x02]
Security level [05]
network state [02] Buffs: 75 / 75
Ep cnt: 3
ep 1 [endpoint enabled, device enabled] nwk [0] profile [0x0104] devId [0x0101] ver [0x01]
    in (server) cluster: 0x0000 (Basic)
    in (server) cluster: 0x0003 (Identify)
    in (server) cluster: 0x0004 (Groups)
    in (server) cluster: 0x0005 (Scenes)
    in (server) cluster: 0x0006 (On/off)
    in (server) cluster: 0x0008 (Level Control)
ep 2 [endpoint enabled, device enabled] nwk [0] profile [0x0104] devId [0x010D] ver [0x01]
    in (server) cluster: 0x0000 (Basic)
    in (server) cluster: 0x0003 (Identify)
    in (server) cluster: 0x0004 (Groups)
    in (server) cluster: 0x0005 (Scenes)
    in (server) cluster: 0x0006 (On/off)
    in (server) cluster: 0x0008 (Level Control)
    in (server) cluster: 0x0300 (Color Control)
    in (server) cluster: 0x1000 (ZLL Commissioning)
ep 242 [endpoint enabled, device enabled] nwk [0] profile [0xA1E0] devId [0x0061] ver [0x01]
    out(client) cluster: 0x0021 (Green Power)
Nwk cnt: 1
nwk 0 [Primary (pro)]
  nodeType [0x02]
  securityProfile [0x05]
Z3LightSoc_PA_010>

````

[image]
[image]

darkxst, 2024-12-16:
This is not really the right place for this. Your probably better off posting this in the forums.

But you will at the very least need a custom built firmware for that chip and you will need to work out pinouts also.

kar200, 2024-12-17:
No worries thank you for the prompt rely. 

I did not realise it was a complex task without having the proper info. I will close this since it's irrelevant.

PS : any chance you can point me to which forum to post this into please?

closed 2024-12-17 by kar200