NabuCasa/silabs-firmware-builder #105: HA offering upgrade to a prerelease firmware, which has "breaking changes" without making it clear

tomchiverton, 2025-04-04:
```
Core 2025.4.0
Supervisor 2025.03.4
Operating System 15.1
Frontend 20250401.0
```
Under Settings > Update it says
```
EmberZNet Zigbee
Installed version
7.1.1.0
Latest version
7.4.4.0
Read release announcement
```

Unless you click that final link, you'd never know this is "pre-release ... this is a breaking change for all firmware " and "this is a work-in-progress and Zigbee software support will arrive over time. "

IMHO this version should not be being offered to end users, and it should be removed from the update service.

puddly, 2025-04-04:
This isn't a pre-release version, nor is it a breaking change. What gives you that impression?

puddly, 2025-04-04:
Oh, I see what you mean. I'll update the release information to clarify the wording. The "breaking change" is for consumers of the SiLabs firmware builder repository and build scripts, not for end-users. Similarly, the referenced command set is provisional but again, this is handled by libraries and transparent to the end-user.

puddly, 2025-04-04:
Clarified: https://github.com/NabuCasa/silabs-firmware-builder/releases/tag/v2024.10.21

closed 2025-04-04 by puddly