NabuCasa/silabs-firmware-builder #109: IKEA TRADFRI devices not responding after update 7.4.4

HFeenstra, 2025-04-05:
All of IKEA TRADFRI devices wont work after updating the stick to 7.4.4

I even removed them from ZIgbee integration, rebooted HA added them again to Zigbee still same result, no responding to switch or movement sensor.

ZIgbee devices like Philips Hue keep working.

HFeenstra, 2025-04-05:
Trying right now, took out the stick in https://connectzbt1.home-assistant.io/firmware-update but still 7.4.4.0 build 0 firmware???
So how and where to update to 7.4.4.1 version??

puddly, 2025-04-05:
Updates for the ZBT-1 are handled by Home Assistant. Go to http://homeassistant.local:8123/config/updates, click the "Check for Updates" button on the top right, and it'll show up: 

[image]

HFeenstra, 2025-04-05:
Okay, thank. Was able to update to 7.4.4.1. Via Home Assistant update.

On/Off IKEA TRADFRI switch works now as expected. 
But,..
Round IKEA TRADFRI switch with dimmer function as well as movement sensor of IKEA still doesn’t work. Even when I delete them from Zigbee network re-adding them and trying to get it working again?

Is this also taken into account for this re-release?

puddly, 2025-04-05:
I think you may have a separate issue, there's no known issue with motion sensors, nor can I reproduce it either my IKEA motion sensor.

HFeenstra, 2025-04-05:
Super you respond so quickly!

I’m sorry but I doubt that, its to much of an coincidence that after the firmware 7.4.4.0 and now to .1 version these devices stop responding.
Again I removed them even from the Zigbee network, re-added them, see them connected, but NO respond to movement or hitting a button.

Up till the firmware upgrade all worked fine constantly.

HFeenstra, 2025-04-05:
Did reload of Zigbee integration, no success.

Okay will now also do a full restart of whole instance with integrations and Home Assistant itself.

negative-video, 2025-04-05:
Similar situation here. Once I updated HA and saw the update for my ZBT-1, I installed 7.4.4.0 and had issues with my battery-powered Ikea Zigbee devices. I followed puddly's instructions and manually updated to 7.4.4.1, which after a quick reconfigure, restored functionality to my [TRADFRI On/Off Switch](https://zigbee.blakadder.com/Ikea_E1743.html). 7.4.4.1 did not restore my [Ikea TRADFRI Motion Sensors](https://zigbee.blakadder.com/Ikea_E1745.html) and they are frozen, constantly reporting the occupancy as "detected". I'd be happy to send over the diagnostic logfile if it would help in troubleshooting.

HFeenstra, 2025-04-05:
Glad to hear I’m not the only one. The IKEA TRADFRI motion Sensor and the round On/Off with dimming switch is still not working for me. 
My devices do respond to identify Click but no respond to button or movement

puddly, 2025-04-05:
If you walk past the motion sensors, does their state change? They should be configured to report their status about once an hour, perhaps they were in a triggered state when the update started and did not reset.

Primajin, 2025-04-05:
Small square switches work with this blueprint for me: https://epmatt.github.io/awesome-ha-blueprints/docs/blueprints/controllers/ikea_e1743/

Round remotes work with this blueprint for me: https://epmatt.github.io/awesome-ha-blueprints/docs/blueprints/controllers/ikea_e1524_e1810/ 

That is on FW 7.4.4.1

HFeenstra, 2025-04-05:
> If you walk past the motion sensors, does their state change? They should be configured to report their status about once an hour, perhaps they were in a triggered state when the update started and did not reset.

No that’s the whole case, no state change what so ever. Have several times reconnect the device to Zigbee network via factory reset. 
Nothing changes. Like previous poster I also have no state change from sensor

HFeenstra, 2025-04-05:
> Small square switches work with this blueprint for me: https://epmatt.github.io/awesome-ha-blueprints/docs/blueprints/controllers/ikea_e1743/
>
> Round remotes work with this blueprint for me: [...]

Tried the round remote including this blue print. Of course it didn’t work either. 
There is something wrong with the connection to the stick. Too much a coincidence that at least two device broke after the firmware upgrade!

Hopefully someone has good advice or an new update?

negative-video, 2025-04-05:
> If you walk past the motion sensors, does their state change? They should be configured to report their status about once an hour, perhaps they were in a triggered state when the update started and did not reset.

That was part of my process for "waking up" the devices to allow them to reconfigure. I tried something more extreme by resetting the motion sensors and re-pairing them (after first rebooting my HA server), which seems to have restored occupancy reporting. The logbook is MUCH more chatty than it used to be, though...

[image]

tubit, 2025-04-05:
Hi there.. Same problem here, all older Ikea Swiches and motion sensors stopped working after updating to 7.4.4.0.

I can confirm that the older square on/off ikea devices work again after updating to 7.4.4.1. Need to re-pair my motion sensors though...

puddly, 2025-04-05:
Can you try clicking the "reconfigure" button instead of re-joining them?

HFeenstra, 2025-04-05:
> Can you try clicking the "reconfigure" button instead of re-joining them?

Tried both methods. Both not giving the result of responding to tap or motion. 
Again the devices do appear in overview and when clicking on identify the light responds and n device. 
But no motion or click responses.

Maart3nL, 2025-04-05:
I tried to update to  7.4.4.1 resulting in the blue light on the front left corner of my yellow beeing on all the time and HA not being able to use the radio anymore. and since ZHA does not start all my automations are gone (all not just the ones that use ZHA devices) removing ZHA fixes that fortunately

Jor-Tech, 2025-04-05:
Had same issue here. 

It said to be linked to Custom XNCP protocol commands by puddly · Pull Request #57 · NabuCasa/silabs-firmware-builder

Mentioned by epenet [here](https://community.home-assistant.io/t/skyconnect-zbt-1-firmware-update-version/591136/30) together with a temporary workaround downgrading to the 2024.8.20 build release.

Works: Release Build 2024.8.20 (Gecko SDK v4.4.4) · NabuCasa/silabs-firmware-builder · GitHub
Doesn’t work: Release Build 2024.10.21 (Gecko SDK v4.4.4) · NabuCasa/silabs-firmware-builder · GitHub

Would be great if this could be fixed.

puddly, 2025-04-06:
2024.8.20 should be identical to the 7.4.4.1 build. If you notice a difference, that definitely is a bug.

Jor-Tech, 2025-04-06:
> 2024.8.20 should be identical to the 7.4.4.1 build from earlier today (not the 7.4.4.0 build!). If you notice a difference, that definitely is a bug.

Just went to bed now but will definitely try the new 7.4.4.1 build tomorrow first thing and report back here. Thanks.

Jor-Tech, 2025-04-06:
> 2024.8.20 should be identical to the 7.4.4.1 build from earlier today (not the 7.4.4.0 build!). If you notice a difference, that definitely is a bug.

Correct, just installed the 7.4.4.1 build, tested, and everything (including IKEA motion sensors which previously stopped working) works as expected. 

To be sure I restarted ha and tested again. 
Everything looks great 👍 

Will keep an eye on it but it looks like 7.4.4.1 indeed does the trick.

Thank you @puddly 🙏

ralphbk, 2025-04-06:
How it possible that this firmware was released without first testing it against IKEA devices?

tubit, 2025-04-06:
Unfortunately I can not confirm 7.4.4.1 working again with Ikea motion sensors.

I updated, stopped HA, unplugged the SkyConnect, restarted HA -> no success. The motion sensor is connected and re-paired, re-learned. No success so far.

tubit, 2025-04-06:
> How it possible that this firmware was released without first testing it against IKEA devices?

As unfortunate as it is: This is open source. The maintainer can not test with all available devices on the planet. :( I have some frustration as well, but I guess that's the price to pay if you are not paying for the software & support. :)

tubit, 2025-04-06:
This is the detail view when re-configuring the device: [image]

HFeenstra, 2025-04-06:
> Unfortunately I can not confirm 7.4.4.1 working again with Ikea motion sensors.
> 
> I updated, stopped HA, unplugged the SkyConnect, restarted HA -> no success. The motion sensor is connected and re-paired, re-learned. No success so far.

I have the exact same still no round switch or sensor working. Stick in a different USB and rebooted the whole device. 
Still no luck sensor and round switch still not working.

ralphbk, 2025-04-06:
> As unfortunate as it is: This is open source. The maintainer can not test with all available devices on the planet. :( I have some frustration as well, but I guess that's the price to pay if you are [...]

But to neglect testing with devices from the majority mass-market supplier is unforgivable and non-serious. And this on the HA approved Zigbee dongle! Unbelievable.

Jor-Tech, 2025-04-06:
> I noticed a difference in the two IKEA Tradfri motion sensors I have. 
> The old, bigger one shows no Firmware on detail page of device. This is the one after the firmware upgrade that does not act normal. 
> 
> The newer smaller IKEA Tradfri motion sensor  does show a firmware version on its detail page. This one is acting normally and detects motion. 
> 
> Hope this gives a direction?

Makes sense as I'm only using the newer smaller ones which are all working fine. 

I do have one old one laying around somewhere. I'll try and connect it for you to see how it'll behave.

I also have sone 5 button round switches in use, but they all work fine.

Jor-Tech, 2025-04-06:
> But to neglect testing with devices from the majority mass-market supplier is unforgivable and non-serious. And this on the HA approved Zigbee dongle! Unbelievable.

One dev, zero budget, free firmware. Testing every device—especially IKEA’s entire showroom—is just not realistic.

You're here helping out, which is awesome. That’s what open source is all about!

Appreciate the passion—let’s build, not bash.

HFeenstra, 2025-04-06:
Okay back to the topic:

I reset IKEA round switch E1524 and reconnected it to Zigbee. Took the longest to get it interviewed. 
At end somehow it connected again and still no success. As if the device was still bound before removing? Any way still no success and even restarted Home assistant to see if that would help. Again nothing. 

Still willing to help but a bit frustrating.

Jor-Tech, 2025-04-06:
> Okay back to the topic:
>
> I reset IKEA round switch E1524 and reconnected it to Zigbee. Took the longest to get it interviewed.
> At end somehow it connected again and still no success. As if the [...]

With the 7.4.4.1 build, IKEA Trådfri devices all fully working:

- Round 5 button remote control, Type E1524, Firmware: 0x24040005

- Motion sensor, Type E1745, Firmware: 0x20022623

- Motion sensor, Type E1745, Firmware: 0x24040005

- Motion sensor (old type), Type E1525, Firmware: unknown

Jor-Tech, 2025-04-06:
@HFeenstra not that I doubt your knowledge, but just to double check and be 100% sure. Can you confirm running **7.4.4.1** build by going to the "_Home Assistant Connect ZBT-1_" integration on the integrations page, select the correct hardware devices and confirm that under _device info_  it says "_Firmware: EmberZNet Zigbee **7.4.4.1**_"

I'm running latest Home Assistant (2025.4.1) with this build, Raspberry Pi 4, ZBT-1 (called Sky Connect why I bought it) and using same IKEA device types as you.. I really wonder why we have different behavior.

HFeenstra, 2025-04-06:
Thanks, this is my overview:

- Round 5 button remote E1524, after reconfigure not showing firmware anymore! And not working to act as switch. Is showing in overview
- Motion sensor E1745, with firmware: 0x20022623 is working
- Motion sensor E1525, no longer showing firmware, after several times reconfiguring it is showing in overview but not responding to motion.

Philips Zigbee devices do work normally.

HFeenstra, 2025-04-06:
> [@HFeenstra](https://github.com/HFeenstra) not that I doubt your knowledge, but just to double check and be 100% sure. Can you confirm running **7.4.4.1** build by going to the "_Home Assistant Connect ZBT-1_" integration on the integrations page, select the correct hardware device and confirm that under _device info_ it says "_Firmware: EmberZNet Zigbee **7.4.4.1**_"
> 
> I'm running latest Home Assistant (2025.4.1) with this build, Raspberry Pi 4, ZBT-1 (called Sky Connect why I bought it) and using same IKEA device types as you.. I really wonder why we have different behavior.

Thanks for thinking with me:

In integration I see the EmberZNet Zigbee 7.4.4.1 (@puddly showed the upgrade steps and I executed them) 
I’m running 2025.4.1 on a Home Assistant Blue device, with the Home Assistant Connect ZBT-1 as integration. Looks identical to me!

Jor-Tech, 2025-04-06:
> > [@HFeenstra](https://github.com/HFeenstra) not that I doubt your knowledge, but just to double check and be 100% sure. Can you confirm running **7.4.4.1** build by going to the "_Home Assistant Connect ZBT-1_" integration on the integrations page, select the correct hardware device and confirm that under _device info_ it says "_Firmware: EmberZNet Zigbee **7.4.4.1**_"
> > 
> > I'm running latest Home Assistant (2025.4.1) with this build, Raspberry Pi 4, ZBT-1 (called Sky Connect why I bought it) and using same IKEA device types as you.. I really wonder why we have different behavior.
> 
> Thanks for thinking with me @Jor-Tech:
> 
> In integration I see the EmberZNet Zigbee 7.4.4.1 (@puddly showed the upgrade steps and I executed them) 
> I’m running 2025.4.1 on a Home Assistant Blue device, with the Home Assistant Connect ZBT-1 as integration. Looks identical to me!

Yes, apart from me running a Pi4 and you the Blue, but I can't imagine that it could make the difference. I'm sorry, I'm out of ideas, but will follow this issue thread and help provide input/feedback from my side where needed/possible.

ralphbk, 2025-04-06:
> > > > How it possible that this firmware was released without first testing it against IKEA devices?
> > > 
> > > 
> > > As unfortunate as it is: This is open source. The maintainer can not test with all available devices on the planet. :( I have some frustration as well, but I guess that's the price to pay if you are not paying for the software & support. :)
> > 
> > 
> > But to neglect testing with devices from the majority mass-market supplier is unforgivable and non-serious. And this on the HA approved Zigbee dongle! Unbelievable.
> 
> One dev, zero budget, free firmware. Testing every device—especially IKEA’s entire showroom—is just not realistic.
> 
> He's here helping out, which is awesome. That’s what open source is all about!
> 
> Appreciate the passion—let’s build, not bash.

I appreciate his efforts, but on the other side, I am paying for two locations of Nabu Casa every year. I think we have the right to expect that some of that cash to be used for basic testing of the firmware releases of the officially supported Zigbee hardware. Users of mainstream Zigbee devices (and I have 50+ of them) shouldn't have our systems brought down by untested releases like this.

Jor-Tech, 2025-04-06:
> I appreciate his efforts, but on the other side, I am paying for two locations of Nabu Casa every year. I think we have the right to expect that some of that cash to be used for basic testing of the [...]

Maybe, but complaining doesn't solve anything, you're message already got across.
Now if you're serious about wanting things to improve, channel that energy into useful feedback and testing. That’s how open source moves forward.

puddly, 2025-04-06:
@Jor-Tech and @HFeenstra , could you post an the automation YAML you're using for the motion sensor? I'm not sure why some people have this issue but others do not.

HFeenstra, 2025-04-06:
@puddly sure, but don’t see the connection between those two? _(Since the firmware upgrade 7.4.4.0 and later on 7.4.4.1 the device isn’t responding anymore)_

Below the YAML of one of the automations, the device is recognized in the Home Assistant environment but not acting as a sensor device or as round remote switch…

alias: Turn on bedroom floor light when there is movement
description: ""
triggers:
  - entity_id:
      - binary_sensor.ouders_motion_sensor
    to: "on"
    trigger: state
conditions:
  - condition: sun
    before: sunrise
    after: sunset
actions:
  - data:
      entity_id: light.osram_flex_rgbw_licht
      brightness: 25
      kelvin: 2700
    action: light.turn_on
  - action: timer.start
    data:
      duration: "00:00:12"
    target:
      entity_id: timer.vloerlicht
mode: single

HFeenstra, 2025-04-06:
I feel more and more the urge to completely wipe the Zigbee integration and network and start from scratch. Meanwhile when I now see how the not working devices are not coming up properly or even don’t show all their states I’m a bit holding back.
Starting from scratch would be a no go for now still. Since most devices are working, but just some of the IKEA TRADFRI devices are not acting normally. 
So still have patience and work around situations to get it working for now….

Jor-Tech, 2025-04-06:
Also don't see the relation of the device doesn't respond or change state.. it's not in the automation. 

`alias: Toiletlamp motion sensor
description: ""
mode: single
triggers:
  - type: motion
    device_id: 863d9a3c92d58329862d824cf9b7236d
    entity_id: 2a7bf811dbd0adb0330712044464c3f1
    domain: binary_sensor
    trigger: device
conditions: []
actions:
  - if:
      - condition: time
        after: "05:30:00"
        before: "23:00:00"
    then:
      - metadata: {}
        data:
          transition: 1
          brightness_pct: 100
        target:
          entity_id: light.toiletlamp
        action: light.turn_on
    else:
      - metadata: {}
        data:
          transition: 1
          brightness_pct: 50
        target:
          entity_id: light.toiletlamp
        action: light.turn_on
  - wait_for_trigger:
      - type: no_motion
        device_id: 863d9a3c92d58329862d824cf9b7236d
        entity_id: 2a7bf811dbd0adb0330712044464c3f1
        domain: binary_sensor
        trigger: device
  - target:
      entity_id:
        - light.toiletlamp
    data: {}
    action: light.turn_off
`

Jor-Tech, 2025-04-06:
@HFeenstra as last resort you could maybe try manually re-flashing using the Silicon Labs flasher add-on as explained here:

https://community.home-assistant.io/t/skyconnect-zbt-1-firmware-update-version/591136/40

But replace the correct firmware file url to the latest.

puddly, 2025-04-06:
@HFeenstra to eliminate differences between Core versions, have you tried downgrading the firmware on your stick? Go to https://connectzbt1.home-assistant.io/firmware-update/ and upload the following file to switch it back to the firmware it shipped with https://github.com/NabuCasa/silabs-firmware/raw/refs/heads/main/EmberZNet/NabuCasa_SkyConnect_EZSP_v7.1.1.0_ncp-uart-hw_115200.gbl . Let me know if that helps.

The only "quirk" I'm aware of with older generation IKEA devices is that some communicate by sending group commands instead of unicast commands. This works properly for me with an old IKEA dimmer knob with the 7.4.4.1 firmware.

Jor-Tech, 2025-04-06:
Don't know if it makes af difference but I actually downgraded from the 7.4.4.0 build to the 2024.8.20 build using the Silicon Labs flasher add-on as explained [here](https://community.home-assistant.io/t/skyconnect-zbt-1-firmware-update-version/591136/40).

Then updated from there to 7.4.4.1 using the build in update feature.

HFeenstra, 2025-04-06:
@puddly thanks for the comment on downgrading the controller.
I right away did this. So Skyconnect has now 7.1.1.0

The round 5 button switch I first re-added it to the network as a new device, did not delete it up front, the interview with the device including config went real smooth within 10 sec. With the SkyConnect 7.4.4.1 it took for ever!
That was a sign for me that it might work, it did. So the round 5 button switch is working.

The motion sensor, old version, is still not coming back. Reconfigured the zigbee network, deleted the current device, re-added it, up till now no success!
So motion sensor is not working correctly. I do see the log files mentioning the firmware is turned off and then right away also mentioned that it dropped the firmware. So really odd!?!

ABEIDO, 2025-04-06:
Having  ongoing issues TRADFRI on/off switch, also had some strange temporary issues with other IKEA devices but i have them working again. 

TRADFRI on/off switch can be readded ok to ZHA but no events is listed when monitoring zha_event in dev tools and thereby it not working.

Have updated to latest fw 7.4.4.1 and reset the device several times also rebooted HA.

Am i correct that the only fix as of now is to downgrade the skyconnect fw to version previous to 7.4.4.0 to get it to work?

puddly, 2025-04-06:
For the switch, have you tried clicking the Reconfigure button on the device page?

ABEIDO, 2025-04-06:
yes i have with no luck, tried again now and it failed again

ABEIDO, 2025-04-06:
Rebooted and redid all again plus the reconfifured again and even if it said it failed it started to work after a couple of minutes.

puddly, 2025-04-07:
For those still affected by this problem and interested in trying a fix, try this firmware out [skyconnect_zigbee_ncp_7.4.4.2.gbl.zip](https://github.com/user-attachments/files/19626391/skyconnect_zigbee_ncp_7.4.4.2.gbl.zip). Unzip the file first and upload the GBL via the web flasher: https://connectzbt1.home-assistant.io/firmware-update/.

You don't need to shut down ZHA to try it (or revert back), you can just unplug the stick and plug it back in.

tubit, 2025-04-07:
@puddly I just flashed 7.4.4.2 and can confirm motion Tradfri sensors working again.

Thanks for your support & efforts!

tubit, 2025-04-07:
> [@tubit](https://github.com/tubit) to confirm, 7.4.4.1 didn't work, but the firmware I posted above does?

yes, 7.4.4.1 did not work but the posted firmware worked. I flashed it via the cmdline python tool. After that, a newly added motion sensor worked immediately.

puddly, 2025-04-07:
For those still having issues with older IKEA devices (or if you were able to reproduce the problem in the past) please give the following firmwares a try and let me know if the problems are resolved: https://github.com/NabuCasa/silabs-firmware-builder/pull/112#issuecomment-2784578073.

This is different from the 7.4.4.2 firmware I posted above but should fundamentally have the same behavior.

HFeenstra, 2025-04-08:
Sorry update on the upgrade. It did work a few times. Since this morning it doesn’t anymore. I do see the events being triggered for movement but no activating the automation. So it won’t tell me that the sensor sensed a motion and the automation kicks off. Still have 7.4.4.2 on the sticks. Might test later back to 7.1.1.0 to see whether that will act as aspected before.

puddly, 2025-04-08:
To confirm, you tested the firmware I linked to yesterday? Or the one from a few days ago?

HFeenstra, 2025-04-08:
@puddly yes the latest 7.4.4.2 version. 

I have the same thought it worked at first and now not anymore. Restarted Zigbee network, reconnected the old ikea Trådfri sensor.

HFeenstra, 2025-04-08:
@puddly sorry for the hassle 
I can now confirm, it was indeed automation related. 
I was too much focused on the device not working that I forgot to check the behaviour of the automation itself!

So my previous comment is no longer valid. Glad you got me focused on the automation!

puddly, 2025-04-08:
Thanks for the feedback! I think this issue can be considered resolved then. The test firmware I linked to above (https://github.com/NabuCasa/silabs-firmware-builder/pull/112#issuecomment-2784578073) behaves identically to the one you tested earlier and should be good to go.

closed 2025-04-08 by puddly

Peter376, 2025-06-04:
Hi all,

A general question: is there anything against _not updating_ the EmberZNet firmware at all?

Reason I ask:
I am still on 7.1.1.0 and my HA system is running okay. I have found multiple forum threads about people having issues with either 7.4.4.0 or 7.4.4.1 and it gives me the jitters.

Thanks!
Pete