NabuCasa/silabs-firmware-builder #110: updating firmware homeassistant yellow

dominicmcglinn, 2025-04-06:
Getting error trying to upgrade
Home Assistant Yellow Radio firmware
10 minutes ago
Update available
EmberZNet Zigbee
Installed version
7.4.4.0
Latest version
7.4.4.1
[Read release announcement](https://github.com/NabuCasa/silabs-firmware-builder/releases/tag/v2025.04.04-1)

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 1:52:13 PM (3 occurrences)
Last logged: 1:53:33 PM

[546674282928] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 300, in async_install
    await flasher.enter_bootloader()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 272, in enter_bootloader
    await self.probe_app_type()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 118, in probe_gecko_bootloader
    async with self._connect_gecko_bootloader(baudrate) as gecko:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/common.py", line 136, in connect_protocol
    _, protocol = await zigpy.serial.create_serial_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/serial.py", line 137, in create_serial_connection
    raise exc.__context__ from None
  File "/usr/local/lib/python3.13/site-packages/serial/serialposix.py", line 385, in _reconfigure_port
    fcntl.flock(self.fd, fcntl.LOCK_EX | fcntl.LOCK_NB)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
BlockingIOError: [Errno 11] Resource temporarily unavailable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 307, in async_install
    raise HomeAssistantError("Failed to flash firmware") from err
homeassistant.exceptions.HomeAssistantError: Failed to flash firmware

Maart3nL, 2025-04-06:
I tried to update my yellow zigbee radio to 7.4.4.1 (was already on 7.4.4.0) resulting in the blue light on the front left corner of my yellow beeing on all the time and HA not being able to use the radio anymore. Zha cant start anymore and the blue light is permanently on.

dominicmcglinn, 2025-04-06:
Worked for me now.

closed 2025-04-06 by dominicmcglinn

jabdoa2, 2025-04-06:
How did you fix this?