NabuCasa/silabs-firmware-builder #111: Update to 7.4.4.1 - Failed to flash firmware

jabdoa2, 2025-04-06:
I get this error:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 300, in async_install
    await flasher.enter_bootloader()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 272, in enter_bootloader
    await self.probe_app_type()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 118, in probe_gecko_bootloader
    async with self._connect_gecko_bootloader(baudrate) as gecko:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/common.py", line 136, in connect_protocol
    _, protocol = await zigpy.serial.create_serial_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/serial.py", line 137, in create_serial_connection
    raise exc.__context__ from None
  File "/usr/local/lib/python3.13/site-packages/serial/serialposix.py", line 385, in _reconfigure_port
    fcntl.flock(self.fd, fcntl.LOCK_EX | fcntl.LOCK_NB)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
BlockingIOError: [Errno 11] Resource temporarily unavailable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 307, in async_install
    raise HomeAssistantError("Failed to flash firmware") from err
homeassistant.exceptions.HomeAssistantError: Failed to flash firmware
```

I am using HAOS on a RPI4. The previous update to 7.4.4.0 worked fine yesterday.

jabdoa2, 2025-04-06:
Sees to be similar to #110. I do not understand how that one has been resolved though.

puddly, 2025-04-06:
How are you running Home Assistant? Was ZHA running before you started the update?

jabdoa2, 2025-04-06:
> How are you running Home Assistant? Was ZHA running before you started the update?

ZHA has been working before. It also still worked fine after the 7.4.4.0 update. ZHA seems to be unavailable now.

I run a standard Home Assistant Operating System on a RPI4 (2GB version). Only other connected hardware is a HM-CFG-USB for Homematic BidCos.

jabdoa2, 2025-04-06:
I restarted HAOS and now ZHA seems to hang with this:
```
 Waiting on integrations to complete setup: {('zha', 'd119b5ca979af4fd7376bbc3022aa28b'): 251580.75539527} 
```
Probably not helpful but HA is not longer starting fully.

jabdoa2, 2025-04-06:
Restarting HA did not help. Rebooting HAOS fixed it. Not sure why this happened but it works again.

bcosynot, 2025-04-06:
Restarting and rebooting HA did not work for me. I’m running the OS on a raspberry pi 3

go2tom42, 2025-04-07:
Same error for me too

MarkVanVenrooij, 2025-04-07:
I had the same error this afternoon after trying to update the radio firmware. I got it fixed by flashing the radio firmware using this [add-on](https://github.com/home-assistant/addons/tree/master/silabs_flasher) although this add-on is experimental it worked on my HA yellow

greglamb, 2025-04-08:
Same for me - I'm on a Green

daveotterness, 2025-04-10:
I am getting the same error on my Yellow, debug log below

`Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 307, in async_install
    raise HomeAssistantError("Failed to flash firmware") from err
homeassistant.exceptions.HomeAssistantError: Failed to flash firmware
2025-04-10 12:04:45.923 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [546483873264] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 300, in async_install
    await flasher.enter_bootloader()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 272, in enter_bootloader
    await self.probe_app_type()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 118, in probe_gecko_bootloader
    async with self._connect_gecko_bootloader(baudrate) as gecko:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/common.py", line 136, in connect_protocol
    _, protocol = await zigpy.serial.create_serial_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/serial.py", line 137, in create_serial_connection
    raise exc.__context__ from None
  File "/usr/local/lib/python3.13/site-packages/serial/serialposix.py", line 385, in _reconfigure_port
    fcntl.flock(self.fd, fcntl.LOCK_EX | fcntl.LOCK_NB)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
BlockingIOError: [Errno 11] Resource temporarily unavailable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 307, in async_install
    raise HomeAssistantError("Failed to flash firmware") from err
homeassistant.exceptions.HomeAssistantError: Failed to flash firmware
`

puddly, 2025-04-10:
@daveotterness Did you enable debug logging as I mentioned above? For the Yellow, you'd do the same for the Yellow integration. The traceback alone unfortunately doesn't have much information.

daveotterness, 2025-04-10:
I did but as it was my first time ever turning debugging on, I may have
done it incorrectly. I will try again, thank you for the reply
-- sent via mobile --

puddly, 2025-04-10:
Ah, I forgot to mention that to download the debug log you need to click the "disable debug logging" button. It will happen automatically after that.

daveotterness, 2025-04-10:
> Ah, I forgot to mention that to download the debug log you need to click the "disable debug logging" button. It will happen automatically after that.

I see some additional lines in this log, INFO DEBUG and ERROR.  Is this what you were looking for? 

`Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 307, in async_install
    raise HomeAssistantError("Failed to flash firmware") from err
homeassistant.exceptions.HomeAssistantError: Failed to flash firmware
2025-04-10 12:56:16.491 INFO (MainThread) [universal_silabs_flasher.flasher] Triggering yellow bootloader
2025-04-10 12:56:16.492 DEBUG (SyncWorker_5) [universal_silabs_flasher.gpio] Sending GPIO pattern GpioPattern(pins={24: True, 25: True}, delay_after=0.1)
2025-04-10 12:56:16.595 DEBUG (SyncWorker_5) [universal_silabs_flasher.gpio] Sending GPIO pattern GpioPattern(pins={24: False, 25: False}, delay_after=0.1)
2025-04-10 12:56:16.697 DEBUG (SyncWorker_5) [universal_silabs_flasher.gpio] Sending GPIO pattern GpioPattern(pins={24: False, 25: True}, delay_after=0.1)
2025-04-10 12:56:16.798 DEBUG (SyncWorker_5) [universal_silabs_flasher.gpio] Sending GPIO pattern GpioPattern(pins={24: True, 25: True}, delay_after=0.0)
2025-04-10 12:56:16.801 INFO (MainThread) [universal_silabs_flasher.flasher] Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-04-10 12:56:16.801 DEBUG (MainThread) [zigpy.serial] Opening a serial connection to '/dev/ttyAMA1' (baudrate=115200, xonxoff=False, rtscts=False)
2025-04-10 12:56:16.807 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [546593179536] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 300, in async_install
    await flasher.enter_bootloader()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 272, in enter_bootloader
    await self.probe_app_type()
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 118, in probe_gecko_bootloader
    async with self._connect_gecko_bootloader(baudrate) as gecko:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/universal_silabs_flasher/common.py", line 136, in connect_protocol
    _, protocol = await zigpy.serial.create_serial_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/serial.py", line 137, in create_serial_connection
    raise exc.__context__ from None
  File "/usr/local/lib/python3.13/site-packages/serial/serialposix.py", line 385, in _reconfigure_port
    fcntl.flock(self.fd, fcntl.LOCK_EX | fcntl.LOCK_NB)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
BlockingIOError: [Errno 11] Resource temporarily unavailable

The above exception was the direct cause of the following exception:
[... 7 lines cut ...]
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
[... cut ...]
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 307, in async_install
    raise HomeAssistantError("Failed to flash firmware") from err
homeassistant.exceptions.HomeAssistantError: Failed to flash firmware
`

CrallH, 2025-04-10:
Running my setup in proxmox, can't install update either. Not with the addon or the integration.

[error_log-12.zip](https://github.com/user-attachments/files/19689853/error_log-12.zip)

daveotterness, 2025-04-10:
Zipped Core log sent via email

puddly, 2025-04-10:
Thanks! From your log, I'm able to see that you have ZHA enabled but I don't see any logging from ZHA happening. Is logging for ZHA disabled or is the integration not running? If it isn't, do you have it disabled or is it in a failure state/retrying setup?

daveotterness, 2025-04-10:
Good catch, ZHA is showing "Failed to set up" .  I do use ZHA as it
controls my smart blinds.  I know this was functioning as of this previous
weekend, and I have not touched any config or really anything at all since
then.   Shall I try turning on debug logging for ZHA as well and rebooting
my Yellow?  Let me know what you think I should do if you think I can help.

[image: image.png]

puddly, 2025-04-10:
I think that ZHA setting up may be interfering with the firmware flashing. If you disable the integration (you can re-enable it afterwards), are you able to upgrade?

daveotterness, 2025-04-10:
That worked!

1) Disabled ZHA integration
2) Rebooted Yellow
3) Upgraded "Home Assistant Yellow Radio" firmware; Successfully (was giving error previously)
4) Enabled ZHA integration; Initialized without error

Thank you @puddly

puddly, 2025-04-10:
For other people with this problem, do you also have a ZHA integration in a failed state?

MarkVanVenrooij, 2025-04-11:
At the moment it was broken zha was in a failed state.  The failed state basically said something along the lines of cannot connect to the radio. Before the first install of the firmware zha was working. 

If I remember correctly I did try to disable it and then try to install again. That didn’t work then. Don’t remember though if I did a reboot in between. 

Hope this helps.

CrallH, 2025-04-11:
I have tried with zha enabled and disabled, makes no difference for me unfortunately.

jabdoa2, 2025-04-11:
Only ZHA. Nothing else using serial.

bcosynot, 2025-04-11:
I had to disable quite a few integrations and add-ons for the flashing to be finally successful. I had already tried disabling ZHA, MQTT, Matter, etc. I tried flashing after disabling other integrations one after another and it eventually worked.

puddly, 2025-04-11:
Which integrations did you disable, @bcosynot? Nothing should be trying to talk to your Zigbee radio other than the integration that is exclusively controlling it, which for now is assumed to be ZHA.

CrallH, 2025-04-12:
> [@CrallH](https://github.com/CrallH) (and anyone else running into this issue) what software are you running that could be interfacing with the radio?
> 
> * Silicon Labs Multiprotocol
> * OpenThread Border Router
> * RaspberryMatic
> * Zigbee2MQTT
> * anything else that uses serial ports

Only zha. Nothing else.

CrallH, 2025-04-12:
I manage to get it updated now. After disabling zha I had to restart HA to get the update to work. 
So, disable zha, reboot, update, enable zha, Victory!!

MarkVanVenrooij, 2025-04-12:
Only ZHA in my case.

jmudes, 2025-04-26:
> I manage to get it updated now. After disabling zha I had to restart HA to get the update to work. So, disable zha, reboot, update, enable zha, Victory!!

Thanks this worked for me

geerlingguy, 2025-05-22:
I'm hitting this on my home HA Yellow... no other serial device present.

```
2025-05-21 21:11:50.045 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547311087872] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 250, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/update.py", line 319, in async_install
    raise HomeAssistantError(
        "Failed to probe the firmware after flashing"
    )
homeassistant.exceptions.HomeAssistantError: Failed to probe the firmware after flashing
```

Have tried disabling ZHA, rebooting, updating the firmware... but it still fails each time it tries updating (I've power cycled with ZHA off twice now).

Tried manually enabling ZHA after a reboot as well, but still it fails initialization no matter what...

```
2025-05-21 13:21:46.218 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry ttyAMA1, s/n: n/a for zha
```

geerlingguy, 2025-05-22:
I've installed the Silicon Labs flasher (as mentioned earlier in this thread):

  1. Enable 'Advanced mode' in my user settings page
  2. Go to Settings > Add-ons and search for Silicon Labs Flasher
  3. Install Silicon Labs Flasher
  4. On Silicon Labs Flasher add-on settings page, go to Configuration
  5. Select the device (E.g. `/dev/ttyAMA1` in my Yellow)
  6. Save the Configuration change
  7. Go back to Info, and click 'Start'

But the logs show:

```
s6-rc: info: service universal-silabs-flasher: starting
[21:28:14] INFO: The selected serial port is not a USB device.
[21:28:14] WARNING: No firmware found for the selected device, assuming firmware is installed.
[21:28:14] INFO: universal-silabs-flasher-up script exited with code 0
s6-rc: info: service universal-silabs-flasher successfully started
```

How do I get the flasher to flash the Zigbee radio?

Peter376, 2025-06-04:
Hi all,

A general question: is there anything against _not updating_ the EmberZNet firmware? 

Reason I ask:
I am still on 7.1.1.0 and my HA system is running okay. I have found multiple forum threads about people having issues with either 7.4.4.0 or 7.4.4.1 and it gives me the jitters.

Thanks!
Pete

ke-bo80, 2025-06-06:
### Solution that worked for me:
My solution was to disable the ZHA integration, then reboot Home Assistant in Safe Mode. From there, I was able to successfully update the firmware on the SkyConnect dongle.

After that, I rebooted back into normal mode, re-enabled ZHA, and now everything is working again.

My setup:

Installation method: Home Assistant OS

Core: 2025.5.3

Supervisor: 2025.05.3

Operating System: 15.2

Frontend: 20250516.0

Hosted on: Proxmox 8.2.7 VM