NabuCasa/silabs-firmware-builder #114: Activity LED disabled after FW upgrade to 7.4.4.1

pjgrun, 2025-04-30:
The blue Zigbee activity LED on the HA Yellow if off after upgrade from 6.10.3.0 to 7.4.4.1.  Under 6.10.3.0 this LED continuously flashed according to Zigbee radio activity.

mukul-k, 2025-05-06:
I have an HA Yellow and didnt know there was a zigbee activity led lol. Can you describe where it is because I can't seem to find it?

mukul-k, 2025-05-06:
> The blue Zigbee activity LED on the HA Yellow if off after upgrade from 6.10.3.0 to 7.4.4.1. Under 6.10.3.0 this LED continuously flashed according to Zigbee radio activity.

I have an HA Yellow and didnt know there was a zigbee activity led lol. Can you describe where it is because I can't seem to find it?

pjgrun, 2025-05-11:
[image]

It is located on the opposite side of the other LEDs, in the area of the Zigbee module.

See also: 

[image]

pjgrun, 2025-09-18:
Upgraded today on HA Yellow to EmberZNet version 7.4.4.2 (just released). Blue LED issue is still present

closed 2025-12-09 by puddly