NabuCasa/silabs-firmware-builder #125: Update to Firmware 7.4.4.2 on HA Green fails

BobbyBlixberg, 2025-09-18:
Here's the related part from the debug log:

```
2025-09-18 10:53:58.861 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [281472261114592] Error during service call to update.install: Failed to flash firmware
2025-09-18 10:57:59.237 ERROR (MainThread) [frontend.js.legacy.202509035] Uncaught error from Electron 21.2.0 on Windows 10
TypeError: Failed to construct 'HTMLElement': This instance is already constructed
construct (src/scoped-custom-element-registry.ts:340:21)
super (src/reactive-element.ts:1037:4)
new c (src/lit-element.ts:130:)
super (node_modules/@awesome.me/webawesome/dist/chunks/chunk.ETAGLSDL.js:19:4)
super (node_modules/@awesome.me/webawesome/dist/chunks/chunk.I5CT6KOH.js:21:4)
I (src/scoped-custom-element-registry.ts:645:4)
J (src/scoped-custom-element-registry.ts:408:8)
importNode (src/lit-html.ts:1218:51)
u (src/lit-html.ts:1639:32)
$ (src/lit-html.ts:1475:11)
2025-09-18 10:57:59.260 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [281472261114592] Error during service call to update.install: Failed to flash firmware
```

ralphbk, 2025-09-18:
I was going to ask if it had been tested on IKEA Zigbee devices this time (unlike 7.4.4.1) but if it doesn't even install, that saves me a worry.

samuelbagforsprivat, 2025-09-18:
Same on HA Blue! Failed to flash firmware, tried several times also with all relevant modules disabled, rebooted between and nothing works.

puddly, 2025-09-18:
Did you previously use it with ZHA?

Jor-Tech, 2025-09-18:
Same thing, it fails immediately.

primedead, 2025-09-19:
I had the same issue. The solution for me was to disable the SkyConnect Multi-PAN in the Zigbee Home Automation integration. I would expect Home assistant would disable every zigbee integration automatically when updating the firmware. It might also help if the errorhandling would inform the user that disabling all zigbee integrations and addons would allow the update to install.

vb7445, 2025-09-19:
Installation method Home Assistant OS
Core 2025.9.3
Supervisor 2025.09.0
Operating System 16.2
Frontend 20250903.5
Zigbee adapter: Home Assistant Connect ZBT-1 (SkyConnect), EmberZNet Zigbee 7.4.4.2
Zigbee integration: ZHA

Before stopping Zigbee2MTTQ-ADON:  Firmware update FAILED
After stopping Zigbee2MTTQ-ADON:  Firmware Update SUCCESSFULL

samuelbagforsprivat, 2025-09-22:
Tryed every possible way of stopping ZHA addon and some other kandidates for problem but nothing works. After failing I can flash back 7.4.4.0 with the Zilicon Labs flasher addon to get it back up working again. Where can I add more logging to get more relevant logs back?

samuelbagforsprivat, 2025-09-24:
Anyone know if 7.4.4.2 will be added to the webflasher? Right now its still on 7.4.4.0

BobbyBlixberg, 2025-10-01:
Same issue for the new FW 7.4.4.3

Please, add the information how to fix this in the description.

vb7445, 2025-10-02:
Had to Stop the Zigbee2MTTQ Add-on and then the firmware, 7.4.4.3 update completed.

enx1998, 2025-10-02:
> Had to Stop the Zigbee2MTTQ Add-on and then the firmware, 7.4.4.3 update completed.

I don't have the Zigbee2MTTQ Add-on installed, but I have the issue installing the dirmware. Maybe this is not the only thing to try to stop before to do the update?