NabuCasa/silabs-firmware-builder #126: Upgrade from 7.4.4.1 to 7.4.4.2 badly failed :-(

tlorent79, 2025-09-18:
Hi, 

I've a HA yellow with CM5.

This morning, I've got invite to upgrade to 7.4.4.2 but then the upgrade blocked at 73% for hours. 

After a reboot, every attempt to upgrade failed again. 
In the mean time, I've lost all connection with all my Zigbee devices. ZHA was continuously 'initialization' then 'failed' then again.

Then I tried the add-on approach Silicon Labs Flasher but it doesn't recognize even my radio at all :
[11:59:52] WARNING: No firmware found for the selected device, assuming firmware is installed.

[image]

[image]

Then I've reboot again, same issue

Losing hope, i've removed the ZHA integration, reboot, tried without success upgrade of the firmware via home assitant yellow integration (radio update) en via the addon Silison Labs ... no success.
I've tried to reinstall ZHA taht is offered ... but also got error message 

[image]

[image]

Thus I'm lost and I found no help on the many forums i've explored.

Please help !! 

Thanks 🥇 , 
Thomas

cheezgr8r, 2025-09-18:
Trying disabling zha while you run the update. I had a similar issue with zigbee2mqtt and it updated just fine after I stopped the addon before running the update. Once updated zigbee2mqtt started normally and my devices worked again!

tlorent79, 2025-09-18:
Thank you @cheezgr8r  ! 

> Trying disabling zha while you run the update. I had a similar issue with zigbee2mqtt and it updated just fine after I stopped the addon before running the update. Once updated zigbee2mqtt started normally and my devices worked again!

I've done even beter :-( i've removed the ZHA integration ... and still not working